X1Plus/X1Plus #355: Alternative Boot System

TiverT, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
No

**The Idea**
In the current version of X1Plus, I power up my printer and get asked by the boot loader if I want to boot into X1Plus or if I want to see more boot options. After a few seconds, it boots into x1plus automatically. I think a system that automatically boots into x1plus and only shows me the boot options if I hold down the button on top of the printer, would be more fitting.

**Software Information**
 - Current X1Plus Version: 2.0

jneilliii, 2024-08-16:
Did you check the user manual?

https://github.com/X1Plus/X1Plus/wiki/User-Manual#quick-boot-versions-10-11

jphannifan, 2024-08-16:
Yes the quick boot flag is what you're looking for.

We already have the physical buttons on the printer mapped to an emergency override.

psiberfunk, 2024-08-16:
Please to RTM on quickboot

closed 2024-08-31 by jphannifan