X1Plus/X1Plus #356: im trying to get spool rfid database,

secretman28, 2024-08-16:
did you guy's know how if the rfid dump is pass to the printer?, if yes can we save it and make a data base for all the type of spool

the big picture is to be able to load non bambu spool and make it work as the bambu one, we can clone the tag but not make from scratch, if we can acess to all of the type we probably can make a rfid writer for the type every one need

dont know if you can put in the firmware that it send it to you and make available probably 1 or 2 day od data will get all of them lol

im just asking if its doable with the custom firmware

repvik, 2024-08-18:
https://github.com/Bambu-Research-Group/RFID-Tag-Guide

secretman28, 2024-08-18:
i know, cant get all of them, my question is, did the printer receive all the tag information?
is it possible to save them? 
in the printer fw you see the key we need?

repvik, 2024-08-18:
Printer receives all the tag information. Don't need the printer to save them, they can be read fine without printer.
But the printer will not accept tags that are not signed by the Bambu Lab private key that only Bambu Lab has. That's the key that would be necessary to make other tags.

secretman28, 2024-08-18:
yes the plan is to get oem tag data base and just clone them without moding it , stick the tag on a spool not bambu to make it "compatible" like this the ams can calculate filament left in the roll etc get the right color or type of spool 

we can make a flash tool that u choose the type and color of spool u wana make a tag for then flash a tag for it 

but we need a data base of all the type and color avaiable. fast solution if your cfw can save the tag dump already decrypt or with the 1 key a decript we can have all of them in no time and bump the project to next step

capull0, 2024-08-22:
> Printer receives all the tag information. Don't need the printer to save them, they can be read fine without printer. But the printer will not accept tags that are not signed by the Bambu Lab private key that only Bambu Lab has. That's the key that would be necessary to make other tags.

Maybe finding and replacing the public key with an own one to create a own signature?
Second idea, patch out the validation of the signature.

Is it possible to extract, modify and repack any firmware (even for p1s?)

jphannifan, 2024-08-30:
Several users have been working on reverse engineering the AMS and the p1 and a1. You might find useful info in these channels of the discord:
https://discord.com/channels/1041781310299062282/1201621143740760135
https://discord.com/channels/1041781310299062282/1246758531248422974

non-bin, 2025-01-20:
I don't have access to the discord, could someone give us an update on A1, P1, and AMS reverse engineering?

TimBozeman, 2025-02-14:
Ya, any updates from the grape vine? Inquiring minds would love to hear about the P1S. 

I tried those discord links, but it said they were empty or I don’t have access.