X1Plus/X1Plus #369: x1plusd: GPIO toggle / interrupt support

jwise, 2024-09-21:
GPIOs should be able to report interrupts, and should be able to have actions associated with them.

in the same breath, it would be good to have the gpiokeys turn to x1plusd generic GPIOs, and for them to trigger actions also -- and for the current gpiokeys actions to turn into x1plusd actions like any other (so they can be triggered from expansion board GPIOs)

jphannifan, 2024-09-21:
I'm working on the gpiokeys part of this.. I have a branch published https://discord.com/channels/1041781310299062282/1184978673313534072/1287095265983205386

It's starting to look like `evdev` is actually the better option for async. It has a built in async read function contrary to `fcntl`.

jwise, 2024-09-24:
I am thinking about this some, and I think my plan is:
* `Gpio`-like objects grow a property `on_change`, which the `GpioManager` sets when it connects.  `Gpio`s that are capable of receiving interrupts are responsible for calling the `on_change` method if it is set.
* `Gpio`s also grow a property `polling`, which is a bool that describes whether the `GpioManager` needs to poll the `read()` function on this `Gpio`.
* Then, the `GpioManager` has a function `on_event`, which registers a callback for when any GPIO changes.  (For convenience, it returns a context manager whose `__exit__` method removes the callback, so it can be used in a `with` block.)

Then, `gpiokeys` can register `Gpio`s that know how to `read()` (they just return their most recently seen value), and know how to trigger interrupts.  And some mechanism will have to be added to generate actions based on the interrupts...

jwise, 2024-09-24:
also maybe the `on_event` contextmanager, on its `__enter__`, could return a generator of events that you could use, if you don't pass a function into it.  so you could use it like:

```python
def _handle_gpio_change(gpio, value):
    print(f"button {gpio} changed to {value}!")

x1plus.gpios.on_event({"function": "button"}, _handle_gpio_change)
```

or you could use it like:

```python
with x1plus.gpios.on_event({"function": "button"}) as events:
    async for gpio, value in events:
        print(f"button {gpio} changed to {value}!")
```

useful in different contexts.  one of them means you have a task devoted to it (maybe not a problem if you already have a task handling something, like from in an event), and the other means that you get lightweight callbacks here and there

jphannifan, 2024-09-26:
The approach I took was to register actions for each gpiokeys input type (push button or momentary switch like the door sensor). So far this works, and there's already a `buttons.json` to load and save this config.

There's some redundancy in my code because I haven't figured out the best way to implement GpioManager. I think the way you're suggesting to do event handling might make more sense. I'll clean it up a bit more and do something about the UI too

jwise, 2024-09-29:
ok, this should work.  now need to wire up gpiomanager to the properties subsystem to trigger actions.

closed 2024-10-02 by jwise