X1Plus/X1Plus #372: Add Camera Auto/Manual Exposure control

lightmaster, 2024-09-24:
**Is your feature request related to a problem? Please describe.**
The stock auto exposure is abysmal, with the video being way too dark. If you add more lights, you wind up with darker shadows under the hot end as the printer's software attempts to lower the exposure automatically. At this point, it seems fair to say that Bambu doesn't care enough to fix it, so can X1 Plus fix it?

**Describe the solution you'd like**
A toggle to switch between Auto Exposure and Manual. When in Manual Exposure, show a dial or slider to adjust the exposure level.

**Describe alternatives you've considered**
Deal with it? But then again part of the allure of 3D printing is DIY and fixing stuff, and I wouldn't have installed custom firmware...

**Software Information**
 - Current X1Plus Version: 2.0

**Additional context**
Add any other context or additional information about the feature request here.

jphannifan, 2024-09-24:
I like this idea, and I'll get a PR up for this soon. In the meantime, here are the commands that can be used to make these adjustments:

Disable autoexposure:
`v4l2-ctl --device /dev/video20 --set-ctrl exposure_auto=1`

Set exposure:
`v4l2-ctl --device /dev/video20 --set-ctrl exposure_absolute=5000`

Enable autoexposure:
`v4l2-ctl --device /dev/video20 --set-ctrl exposure_auto=3`

lightmaster, 2024-09-24:
Thanks, will try those in a little bit when I get ready for a new print. Should those commands survive a reboot or multiple prints, or just until the OS goes to change the auto exposure like it normally would?

jphannifan, 2024-09-24:
> Thanks, will try those in a little bit when I get ready for a new print. Should those commands survive a reboot or multiple prints, or just until the OS goes to change the auto exposure like it normally would?

This isn't something I've tested extensively, but the exposure is not reset after a soft reboot. Here's what I tested:
```
v4l2-ctl --device /dev/video20 --set-ctrl exposure_auto=1
v4l2-ctl --device /dev/video20 --set-ctrl exposure_absolute=5000
```
Exposure is set to 5000 and remains at that value after running the `reboot` command.

I have not tested to see what conditions override this or if a power cycle clears the setting, but I'll look into it

lightmaster, 2024-09-24:
My exposure caps out at 2047, can't even set it to 5000. 50 seems like a good value so far. I did notice something interesting though, if you set exposure to manual while the chamber light is on, then exposure_absolute doesn't seem to have any effect. Had to set exposure_auto to manual while the light was off, then set the exposure_absolute and gamma.

Gonna test these settings with a print to see how they look with more than just the bed there: exposure_auto=1, exposure_absolute=50, gamma=120

jphannifan, 2024-09-25:
When I was testing earlier, I was only modifying the exposure. I'll have to double check whether gamma is on a linear or log scale here. As long as I set `exposure_auto=1` beforehand, I have to adjust my exposure to 50000 to get a decent image with my LED strips turned off. With the LED strips on though, anything above 2500 was over-exposed for me.

Also you probably discovered this already but you can also adjust contrast, sharpness, and brightness. This generated a decent image (LED strip turned on)
```
v4l2-ctl --device /dev/video20 --set-ctrl exposure_absolute=200, gamma=120, sharpness=0, brightness=50, contrast=40
```

lightmaster, 2024-09-25:
I've got the top of mine lined with 24V LEDs, and it is really bright in person. If I set my exposure to 200 and brightness to 50, most of the image is blown out. Also, I don't see any change when adjust contrast, 0 and 95 look exactly the same.

Mine says `exposure_absolute` has a max of `2047`.

```
[root@BL-P001-sdcard:~]# v4l2-ctl --device /dev/video20 --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : PC Camera: PC Camera
        Bus info         : usb-ffe00000.usb-1
        Driver version   : 4.19.111
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : PC Camera: PC Camera
        Serial           : 01.00.00
        Bus info         : usb-ffe00000.usb-1
        Media version    : 4.19.111
        Hardware revision: 0x00000506 (1286)
        Driver version   : 4.19.111
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : PC Camera: PC Camera
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000013: from remote pad 0x100000a of entity 'Extension 4': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1920, Height 1080
        Default     : Left 0, Top 0, Width 1920, Height 1080
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=56
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=56 value=56
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=120
                           gain 0x00980913 (int)    : min=0 max=128 step=1 default=28 value=35
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=166 value=50
```

jphannifan, 2024-09-25:
hmm mine claims to be 80000.

```
[root@BL-P001-sdcard:~]# v4l2-ctl --device /dev/video20 --all
...
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=34 value=34
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=56 value=56
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=150 value=150
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=32 value=2
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=256 step=1 default=23 value=23
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=80000 step=1 default=312 value=624
```

lightmaster, 2024-09-25:
Contrast, gain, and backlight compensation all have different maxes.