X1Plus/X1Plus #389: X1Plus has unauthenticated TCP adbd running by default

jwise, 2024-10-23:
This is truly silly.  It looks like sec_dbg_ctrl_usb.sh is enabling usb_adb_en, which also turns on network ADB.  Ugh.  We do want to keep USB ADB working, but unauthenticated network ADB is bad -- and it seems like adbd wants to do that no matter what.  Maybe iptables that off?

riptidewave93, 2024-10-27:
We can either iptables it off, or disable it completely; but if we do the latter I feel like adding a UI toggle feature to enable/disable may be a good idea.

Due to the above, I feel like a default iptables block rule (with documentation on how to remove for those who wanna use network adb), is the best option here in the short term.

elderlabs, 2025-01-23:
There's a flaw in the `/etc/init.d/S50usbdevice` service script that prevents termination of the processes spawned. Line 390 in the `program_kill()` function.
```
        P_PID=`ps | grep $1 | grep -v grep | awk '{print $1}'`
```
should be 
```
        P_PID=$(ps -e | grep "$1" | grep -v grep | awk '{print $1}')
```
I've cleaned it up a bit to follow modern bash standards, as it's clear there have been many hands in this file, some brilliantly skilled, others less so. The addition of the `-e` flag properly outputs a list of all running processes, which is then further filtered down to the desired process ID, which is then passed on to the `kill` command.

With this, users are able to "gracefully" terminate all services related to the running adb daemon and the USB interface it runs alongside with `/etc/init.d/S50usbdevice stop`. At a glance, it seems like the adb daemon may only be available over the USB network interface, though I've neither looked nor tested, so I cannot be certain.

errge, 2025-04-24:
I can confirm that this ADB on port 5555 is there, fully open, not just on an USB dongle, but on the default WIFI.

`adb connect ip:5555 ; adb shell` gives root from any wifi connected device without any kind of authentication, lan code or password.

jwise, 2025-05-04:
#478 should fix this.

commit 60049607 (2025-05-05): x1plusd: avoid exposing unauthenticated services (#478)

By default, the root password was root (#387), which meant that even if the sshd wasn't on, the ftpd would allow root access. And by default, there was an adbd running on port 5555, which was also bad (#389). x1plusd controls for both of these now on the way up.

* x1plus.sshd: always set a root password, even if sshd is not enabled (fixes #387)
* x1plusd.sshd: add setting "adb.enabled" that triggers iptables rules for adb (fixes #389)
closed 2025-05-05 by jwise