X1Plus/X1Plus #401: X1Plus not allowing new filaments in AMS, example: TPU For AMS

psiberfunk, 2024-11-22:
**Describe the bug**
Per discord discussions in a couple threads ([here](https://discord.com/channels/1041781310299062282/1184978673313534072/1308799029986983997) and [here](https://discord.com/channels/1041781310299062282/1184979423540285540/1309273080459169833)) , there appears to be an issue when telling the AMS to utilize a newer filament type like "TPU for AMS", there are a couple of issues.  To explain this in detail, let us define a process to add to the discussion:

DEFINE:  **OEM Rinse** as upgrade all the way to **_current_** OEM (1.08.02), update your AMS FW as well while you're there, then disconnect the AMS completely (unplug cable),  then reinstall X1Plus after downgrading all the way to firmware R again, usually through a couple of intermediate steps.  Now you have X1Plus, but with the latest AMS firmware.

### Issue details / Reproduction

1) Assuming you are on the latest X1Plus, having not recently come from latest OEM:   before doing any **OEM Rinse** you simply will be unable to select the new filament type at all from the onscreen menu, presumably because the printer/AMS does not know the new filament types exist at all.  In this situation, attempting to set the proper filament type via Bambu Handy or Bambu Slicer, or by triggering RFID reads will result in the onscreen display briefly displaying "TPU for AMS" before going back to this '?' screen: 
[image]

2.  If you go through the **OEM Rinse** procedure, you will temporarily (for a day or so) have a working setup where you CAN set the value in the onscreen display to TPU-for-AMS and it will stick, without any complaint, at least via RFID reads and via the Bambu Slicer (presumably via MQTT?).  While it's annoying to have to go through the **OEM Rinse**, it at least seemed to be a solution of sorts....   However, if you wait a day or so , possibly by rebooting your printer a couple times as well, something will change, and suddenly the printer will display different behavior again.  However this new behavior is slightly different than in use case 1).  Specifically, now you can attempt to set the new filament type in the AMS screen (e.g. the "AMS-for-TPU option is in the dropdown), but upon trying to confirm it, you get told it's incompatible and it resets back to the '?' setting, as shown in this photo and video attachment:
[image]
Video:
https://github.com/user-attachments/assets/030ca1ff-ff6f-4059-8cf0-408bc00f68fd

**Expected behavior**
The AMS page should behave as in the OEM software and 1) show the new filaments as they arrive (probably by updating to /userdata/upgrade/filament/ota-filament-v01.10.00.27-20241119172919.zip.sig , which appears to be the current one). **and**  2)  Allowing for the selection of the updated filaments  (e.g. updating whatever binary/database/whitelist is causing the "not supported" message in the UX above.

**Screenshots/Photos**
Inline, see above

**External Links**
See discord links above

**Additional Details:**
X1C
 - X1Plus version:  2.0 release

**Is the bug present on OEM firmware?**
No, not present on OEM.

psiberfunk, 2024-11-22:
@jwise this is the glitch described in discord

psiberfunk, 2024-11-22:
I found ""%1" is not supported by AMS." in /opt/x1plus/lib/printer_ui.so  .. so that's at least a starting point as to where the message is coming from, but i suspect that's obvious to people who know the innards.

jwise, 2024-11-22:
We chatted some on Discord.  The "is not supported by AMS" message comes from this codepath:

```
if (filament.index !== 254) {
                    if (nameCombo.currentText.indexOf("TPU") != -1 || (((vendorCombo.currentText == "Bambu Lab") || (vendorCombo.currentText == "Bambu")) && (nameCombo.currentTex
t.indexOf("PET-CF") != -1 || nameCombo.currentText.indexOf("PA6-CF") != -1))) {
                        dialogStack.popupDialog(
                                    "TextConfirm", {
                                        name: "select filament info not supported",
                                        type: TextConfirm.CONFIRM_ONLY,
                                        text: qsTr("\"%1\" is not supported by AMS.").arg(editing.name)
                                    })
                        return
```

Is there an update to FilamentPad that we need to take from a base firmware?

The "shows up and then goes away" thing is a different question.

psiberfunk, 2024-11-22:
Note:  before i tried all this I tried the X1Plus intermediate build that lets you update the AMS firmware alone from X1plus to the latest version and that didn't do the trick to enable anything.   Documenting that for completeness so that people know it's not just a simple AMS firmware upgrade issue.

psiberfunk, 2024-11-25:
Note that before all of this I tried first to just use the build 2.0-17+gd91f2ae.x1p  from the AMS FW update patch, and updating the AMS firmware alone wasn't enough:  https://github.com/X1Plus/X1Plus/actions/runs/11429574206

patschi, 2025-01-02:
The "TPU for AMS" was recently briefly back in stock, so I purchased one as well. So does this mean that this filament can not be used with X1P as of today at all, with no fix/permanent workaround in sight? (Beside the workaround which needs to be applied daily outlined in the original PR)

jwise, 2025-01-03:
I have ideas for how to fix this but do not have any of the affected filament. If anyone who does and can compile a firmware wants ideas for things to test please hit me up on discord. Or send me a roll :)joshua

rwbot, 2025-01-03:
@jwise messaged you on discord

patschi, 2025-01-03:
> I have ideas for how to fix this but do not have any of the affected filament. If anyone who does and can compile a firmware wants ideas for things to test please hit me up on discord. Or send me a roll :)

That sounds good! (Thank you very much for all your effort btw, Joshua! Pre-ordered X1P Expander a few days ago :))

I have purchased TPU for AMS and my first X1C a few days ago. Once they arrive and I get X1P flashed, I can assist with debugging, if needed. Unfortunately the rolls are yet again sold out.

But looks like @rwbot is a quicker help. Thank you all very much!

jwise, 2025-01-03:
Latest ota-filament.zip is https://public-cdn.bblmw.com/upgrade/device/BL-P001/filament/product/ota-filament-v01.10.00.31-20241224192933.zip.sig

myxxmikeyxx, 2025-01-05:
I just got this and tried to install it and sadly getting the same thing. I cant even see the TPU for AMS option. I just see a TPU 93(A / HF) but those wont work either. (Mostly commenting so I can stay up to date on this issue).
Could I use the stock firmware, update it, print my print, and then go back online to bambu sites and downgrade back to a supported X1Plus firmware version? I really just have a print that I need to do and then can go back.

emilbayes, 2025-01-20:
I just installed latest X1Plus coming from latest official firmware. My AMS's were on version 00.00.06.44 which seems to support newer filaments (?) but X1Plus warns me that I should downgrade to 00.00.06.40 which doesn't seem to read some Bambu filaments correctly (eg. PETG HF). Not sure if the AMS version influences RFID reading or decoding filament or if it's all managed by the printer OS?

patschi, 2025-01-21:
I'm not sure if there has been any workaround found during latest chats on Discord. It was mentioned that just updating AMS to 00.00.06.44 is apparently not sufficient and it does still not work. Not sure what it has to do with the `ota-filament.zip` Joshua has provided above...

WimVDK, 2025-01-24:
https://pbs.twimg.com/media/GiEigIZXMAAtZWw?format=jpg&name=large

This just definitely settles it for most of us even more. Do not update!

patschi, 2025-01-24:
I think you posted on the wrong GitHub issue?

myxxmikeyxx, 2025-01-25:
@WimVDK @patschi the filament.zip.sig extracted from the firmware (I think) and is used in the slicers (like orca and Bambu) to know the default print settings for the filament. The problem is that the printer and AMS don’t have the right firmware to know how to read the new RFID tag because it is not in the database on the printer. You can used the filament.zip.sig to look at the settings and manually enter them for the unknown filament. I did this in orca and it warns me about it but it seems to be working. It does say this is an unknown filament are you sure you want to use it but since I used the temps for first layer and set the speed in a custom TPU filament profile from the information in the zip, it works.

PR #477 (2025-05-04): bbl_screen: add support to update filament database from Bambu Lab
In addition to being able to manually override the filament database (for those who might be hacking on the AMS), we add UI to download a recommended version of the firmware database, with some modern Bambu filaments in it.

Fixes #401.

[image]
commit 50f93784 (2025-05-04): bbl_screen: add support to update filament database from Bambu Lab (#477)

In addition to being able to manually override the filament database (for those who might be hacking on the AMS), we add UI to download a recommended version of the firmware database, with some modern Bambu filaments in it.

Fixes #401.
closed 2025-05-04 by jwise