X1Plus/X1Plus #407: X1Plus Device Firmware update broken.

masc2279, 2024-12-07:
**Describe the bug**
Using the latest in the main branch when you go to firmware it shows the AMS have a new version which is expected. Then select the AMS and have it install update. When multiple AMS are present it will only update the first AMS and report completed. even when selecting the any of the other devices and trying to update the firmware it updates only the first one and reports as complete. 

As shown in screen shots below it refuses to update any of the other AMS's.

As well as false firmware dialog is presents and should not be shown saying version 99.00.00.00

**To Reproduce**
Steps to reproduce the behavior:
1. Goto Update firmware view in settings
2. Select one of multiple AMS's and select install new firmware.
3. Wait and see that it completes and reports success.
4. See error. Only the first AMS gets updated the other three AMS's are ignored even if you ask for one of the non updated AMS's it will always flash the first AMS and ignore the rest and report as successful ...

And yes printer was rebooted multiple times and re tried which did not work. 

**Expected behavior**
All connected AMS should be updated not just the first one in the chain. Also false firmware dialog is presents and should not be shown saying version 99.00.00.00

**Screenshots/Photos**

[image]

[image]

[image]

[image]

[image]

[image]

[image]

**Additional Details:**
 - Printer type (X1/X1C/X1E):

[image]

[image]

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
This bug is not present in OEM Version and has been tested on my printer lab. 2 of my 8 X1C's are on X1Plus and they both have the same issue.

heathen711, 2025-01-25:
Adding to this since it's silent: I'm running into this too.

masc2279, 2025-02-01:
In my opinion, this project appears to be dead. The developers have been occupied with developing hardware that will not be permitted to operate due to new terms and conditions.  Furthermore, there has been a complete lack of bug fixes or releases since the initial release on June 24, 2024. 

With Bambu’s recent modification of the terms and conditions, the project will no longer have access to future firmware updates or changes and the creators have not said a single peep about the future of this project.

Sad, since it had so much promise.

WolfwithSword, 2025-02-01:
> In my opinion, this project appears to be dead

@masc2279 Pretty active git history, communication, roadmap, and discord activity for a dead project 🤔 

There's been plenty of bug fixes and PRs unrelated to the upcoming expansion board, most of which is to setup more foundation work for things as well. There just hasn't been a reason to do a full new versioned release *yet* but one is on its way.

As far as the expansion board goes and how it will work with the recent T&C - there is a blog post [specific to the expansion board describing its lack of impact](https://www.crowdsupply.com/accelerated-tech/x1plus-expander/updates/bambu-labs-security-updates). Also, a fair bit of work being done in relation to the expansion board is directly relevant as well to non-expansion codebase, such as modularizing various systems within x1plusd, all relevant to the base x1plus. 

The project's still pretty active, with plenty of changes lined up for the next version. If you'd like to hasten updates, you're welcome to volunteer and contribute your own bug fixes.

masc2279, 2025-02-01:
@WolfwithSword  Hello WolfwithSword,

I hope you are correct that there is forward movement. However, the "Pretty active git history, communication, roadmap, and discord activity for a dead project"  as you state means nothing if you don't have the backing of Bambu nor if you don't produce public updates.

psiberfunk, 2025-02-01:
Have you maybe joined the discord ?  There's tons of active discussion there.  I think your view here is borne of a lack of current information and understanding.  We don't need Bambu's permission to continue to provide updates.  Additionally the X1 series firmware is mostly feature complete outside the occasional filament type addition and such.  That some unfunded developers are a couple months out of date really isn't a mark of a "dead" project.

jwise, 2025-02-02:
@masc2279 Hello Scott,

I've been actively writing software for X1Plus recently, of course in addition to working on Expander.  I have posted public updates as to the future of X1Plus in terms of Bambu's most recent updates, which contradict some of the things that you have posted in this thread; I'm not sure you're all that interested in actually reading them, though (presumably, if you were, you already would have?), so I hope you'll forgive me if I don't take the time to hyperlink it here.

I'm sorry to hear that you find that the best use of your time right now is to post negativity in response to active contributors to an open source project.  Whatever the circumstances are in your life that lead this to be the case, I truly and genuinely hope they improve for you; it is trying out there for many of us, and if that includes you, you're not alone.  But for the time being, since these comments are not productive and do not meet the standards I want to set for how this community will run, I've removed your ability to comment in this repository.  Feel free to send me mail if you have any questions.

psiberfunk, 2025-02-02:
P.S.  I have also noticed the root problem/bug in this thread, along with others who have multiple AMSes.  I solved it accidentally by going back to OEM to upgrade my second AMS, but that's a pain in the butt workaround and this bug itself is still valid IMO.

masc2279-return, 2025-02-02:
@jwise Hello,

It is intriguing how individuals tend to dismiss or criticize conversations and people that present challenges rather than engaging in constructive dialogue. When such attempts fail, they resort to censoring the individual involved. A more effective approach would have been to establish a connection between the conversation and facilitate a constructive dialogue based on the potential validity of the perceived issues, which could have been substantiated through evidence or facts. Instead, the decision was made to resort to criticism and censor ship ... 

Just sad.

You do know its a futal process when you can create unlimited accounts right ?

masc2279-return, 2025-02-02:
@psiberfunk 

Thank you ... you are correct. This solved the issue. And also correct that its still a valid bug that all multi AMS users will face.

masc2279-return, 2025-02-02:
@heathen711 

see post from @psiberfunk , The solution worked.

heathen711, 2025-02-03:
I went back to stock but handy didn’t offer that version of AMS. I could try again, but that still doesn’t resolve the problem, it’s simply a workaround.

Also if in the future the rolling back is harder, it’s just going to be a PITA. So I would rather wait for a proper fix here, so I’ll sit where I am with the mixed AMS versions.

Luckily I haven’t noticed any issues, but my AMSs are only slight above the recommended and have not noticed any issues as of yet printing.

If anyone comes up with a patch, I have the setup to test it with too ;)

I do QA for software for a living, I’m comfortable living with bugs…

emlynmac, 2025-06-19:
I just discovered this issue, after updating to the v3 firmware release.

3D-Pink-Girl, 2025-06-20:
Same issue with my second ams not updating to 06.44. First ams is updated. This issue appeared after updating X1plus to 3.0

dvickers79, 2025-08-05:
> Any predictions on 3.1 release. I'm a little hesitant on doing a print on the X1 with mismatched firmware between AMS modules

I will say I've done dozens of prints since the update and with mismatched firmware...no problems that I've seen, for what it's worth.

s4y, 2025-09-08:
> This will be fixed in X1Plus 3.1.

I am running into this exact issue in X1Plus 3.1, can I provide any helpful information?

emlynmac, 2025-09-08:
Confirmed; fixed in 3.1.

Thanks!