X1Plus/X1Plus #418: Optionally disable AMS runout purge

Renegade605, 2025-01-03:
**Is your feature request related to a problem? Please describe.**
When the AMS spool runs out (ie. The end makes it past the first feeder gear) the print can continue because the extruder gear can still advance and retract the filament. 

However, if the print finishes or calls to switch to a different spool, there is no longer any way to pull the filament all the way back and on to the spool. The default behaviour is therefore to purge all the filament in the system down the poop chute. This can be a significant amount depending on where the AMS unit in question is installed. 

In some cases, it is more desirable to save that filament than it is to return to a "safe" state. 

**Describe the solution you'd like**
A selectable setting to disable this behaviour, ideally with three options:
- Purge Normally
- Disable After Print Completion
- Disable Always

When disabled, show an alert box on the built-in LCD that AMS spool is empty and the filament needs to be removed.

The user can then manually cut and remove the remaining filament before clicking a confirm button, and the printer will continue as normal. 

Alternatively, the user could click a different button to purge the filament this one time, without changing the setting. 

**Describe alternatives you've considered**
As far as I can tell, avoiding the purge mid-print is impossible. If the print has completed, the user can cut power to the printer to stop the purge. However, this requires being present and attentive at the right time to intervene. 

**Software Information**
 - Current X1Plus Version: 2.0
 - Base Firmware Version: 1.08.00.00
 - AMS Version: HW AMS08 / FW 00.00.06.40

**Additional context**
In this case, I was batch printing new spools and I don't particularly care what color they are. I made it through the print at almost the exact time the spool ran out, resulting in a purge of somewhere around 1.5m of filament. I would have much preferred to start the next printed spool with the little bit of white that was left and let it mix in to whatever color I was following it with, rather than turning it into a useless blob that goes straight into the garbage.

xorinzor, 2025-01-14:
Unless the software is able to track the filament very accurately it would be difficult to apply the correct settings per filament without the sensor detecting a switch between the filaments. I believe that is partially why the purge happens too, to switch to a known-state before resuming the print.

Renegade605, 2025-01-19:
> Unless the software is able to track the filament very accurately it would be difficult to apply the correct settings per filament without the sensor detecting a switch between the filaments. I [...]

I'm not sure what you mean and I think you misunderstood what I'm talking about.

24c, 2025-01-21:
I get it @Renegade605 I find the filament run out logic even mid print frustrating, as even if I feed another filament in the now empty butbsame spool feeder, aka butt the filament up to each other, it invariably toggles between Orca and the X1C with pop up error/dialog boxes that repeat themselves.