X1Plus/X1Plus #423: Chamber heat control!

markaudacity, 2025-01-17:
The X1C doesn't have a chamber heater, but turning the bed heater to max and running the aux fan does a respectable job of heating up the build chamber.  It's possible to accomplish this with a timer in the startup Gcode, but since Gcode doesn't allow loops, there's no way to get feedback on how quickly or slowly the chamber gets up to temp.

Would it be possible to implement M141 (set chamber heat) and M191 (wait for chamber heat) to do this automatically and without over-undershoot? I have Gcode that works to do the preheating, with a very hacky way of getting it to check chamber temp at various times to see if the temperature is up to one of a set of hard-coded values, but it's ugly, inflexible, requires knowledge of Gcode to edit, and doesn't allow setting specific temperatures from the slicer.

It won't turn an X1C into an X1E, but it might make an X1D out of it. ;)

Gugarutz, 2025-01-24:
I have been looking for something like this for half a year almost. would be very good to have

reipet, 2025-02-15:
I know some activities in this direction, but I would also add the possibility to control an external heater. I have installed one with its own Control Panel, but much nicer would be an rpi to control several addon function for the printer via MQTT maybe. Like another Webcam, Lights Control, Ext. Blower fan control etc.