X1Plus/X1Plus #425: Thoughts/plans on the Autorizathion control update

Pastitas, 2025-01-19:
https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2
Just wanted to make sure you've heard about this and want to know if you know of any changes are coming to the x1plus installability etc,

Aasikki, 2025-01-19:
I hope the old method of jailbreaking is made available again. Imi it's critical to have an option to install this on a printer that's never been online, without ever bringing it online. People working with nda's and their customers 3D models absolutely can't risk connecting a bambu printer online, and I imagine those people would be really happy to be able to use X1Plus to get firmware updates and connect their bambus to lan without the fear of leaking their gcode or stl files.

hibatos, 2025-01-19:
It seems that installing the X1P firmware is not currently possible, the [third party firmware](https://bambulab.com/de-de/third-party-firmware/plan) page is broken and redirects in a loop.
If that is the case than we desperately need a new way of jailbreaking the bambulab printers.

candre23, 2025-01-19:
Can confirm.  Ability to opt into the 3rd party FW has been disabled.  Anybody who hasn't already gone that route no longer has the option.

Flare0, 2025-01-19:
It worked fine for me. I just went through the process to opt in to the 3rd party FW, downgrade firmware and install X1Plus without any issues.

jheyneman, 2025-01-19:
> It worked fine for me. I just went through the process to opt in to the 3rd party FW, downgrade firmware and install X1Plus without any issues.

It's not working for me either, just loops.  I tried multiple browsers just in case.

candre23, 2025-01-19:
After about two dozen attempts on both the DE and US sites, I was finally able to opt in.

Still, this demonstrates the absolute need for a root option which does not rely on the good graces of BL, as they have made it abundantly clear they have none.

hibatos, 2025-01-19:
It could also very well be a load problem, caused by the boat load of users that wanted to opt out before this update gets forcefully rolled out.

psiberfunk, 2025-01-19:
The process does still work folks, just be patient, it appears to be a load issue.

quartzar, 2025-01-19:
I can confirm it does NOT work for me on desktop (Firefox, Kubuntu 24.04), including on an incognito tab.

However, thanks to the comment from @jheyneman , **I am able to reach the enrolment screen from my mobile phone** using Firefox.

vmdude, 2025-01-19:
I confirm I've been able to join 3rd party firmware plan through Firefox on iOS as well, but it's a d*ck move from bambulab IMHO

arkku, 2025-01-19:
I was able to join the 3rd party firmware plan today. I also got the loop, but it seems that if you wait maybe 30 seconds on each page before clicking anything, it will work. There is probably a hidden automatic captcha that thinks you are a bot if you act faster than a snail.

docrobert8404, 2025-01-19:
I was able to join, BUT, my printer won't connect to my account, at that interface it just won't refresh no matter what region I choose. Grabbing a new router tomorrow as all the extra ones I have are junk so I can finish the troubleshooting guide. I have a new AP board and it says the firmware version is all 0s, I have no idea if I can load X1Plus as it is or what firmware version it really is. Any ideas?

arkku, 2025-01-20:
I had an ad blocker enabled on my Safari (on macOS), it did not prevent doing this. The only thing was that I had to wait a while before each of the three actions, or it would fail and loop back to the beginning through a cloudflare captcha.

Snowrisk, 2025-01-20:
Theres an update from Bambulab:

https://blog.bambulab.com/updates-and-third-party-integration-with-bambu-connect/

SgtAstro123, 2025-01-21:
I am a new user of x1plus.  I had no issues following the download procedure.  If the Bambu Labs Firmware site wasn't working for you, it was most likely a load issue as people has said already... or they quickly backtracked from going closed source after the backlash they got from the community.

ForsakenHarmony, 2025-01-21:
> Theres an update from Bambulab:
> 
> https://blog.bambulab.com/updates-and-third-party-integration-with-bambu-connect/

@Snowrisk yeah it still requires bambu connect for orca slicer and they're kinda gaslighting their customers

See Louis Rossmann's Video: https://youtu.be/W6MybDJfmmY

Snowrisk, 2025-01-22:
> [@Snowrisk](https://github.com/Snowrisk) yeah it still requires bambu connect for orca slicer and they're kinda gaslighting their customers
> 
> See Louis Rossmann's Video: https://youtu.be/W6MybDJfmmY

@ForsakenHarmony 
I had understood that the “Developer Mode (Optional)” does not require BambuConnect.

Snowrisk, 2025-01-23:
@ForsakenHarmony Oh, I haven't seen that one yet....

zviratko, 2025-01-24:
> But last comments (see orca issue) seems look like you need still an online access for certificates required in dev mode to be renewed or via sd card, which is a pain ... they could just implement a offline key exchange as the video above also suggests and use it also for the cloud.. this way the user has the control over the keys and not BBL.

They claimed that you can use Bambu Connect completely offline so I don't think it will follow the usual PKI hierarchy but instead just pin the certificate or keys to the printer.

psiberfunk, 2025-01-24:
> They claimed that you can use Bambu Connect completely offline so I don't think it will follow the usual PKI hierarchy but instead just pin the certificate or keys to the printer.

There's a bold implicit assumption here that they'll do it right :).

zviratko, 2025-01-24:
I'm not sure I'd call it "right". If what they want is security, then the solution is not to embed a certificate in Bambu Connect but use some sort of short-lived tokens (or short lived certificates that _are_ part of PKI hierarchy, even if the printer or something generated in their cloud is the root). Maybe they are doing that, I haven't looked at it much yet. 
Of course they could just make it "simple" and use the certificate just as a way to transport the keys around, that is not unusual, but that's harder to manage correctly (how do you revoke/remove an old key, when? Short validity takes care of this).

I would go as far and say that the only robust enough security solution to what they indicate are the problems is some sort of "Bambu Authenticator" app on the phone (maybe they are making that part of Bambu Handy) that would sign/authorize-signing-of all gcode going to the printer, but that' completely unrealistic because if you keep even simple things like moving the toolhead arround with buttons in the Studio go without authorization then bad actors can loop that a 1000 times and destroy your printer in a moment and you get the blame for not securing it enough. But remove that functionality and it would cause another s*tstorm amplified by all haters. You can't win without locking things down too much.

So let's see how that evolves first. I expect Bambu Connect in its general release version to be quite different from what it is today And if bad actors are waging wars online trying to destroy printers and manufacturer's reputation, then let's hope Bambu isn't the one hit first when in fact they are the only ones even talking about security(believe them or not).

psiberfunk, 2025-01-24:
> I'm not sure I'd call it "right". 
By right I meant "actually secure" rather than any moral judgement on what they're doing.  In any case, Orca has (sanely IMO) just rejected Bambu's proposed change as-is, telling Bambu exactly how much their ideas area appreciated by the community.  I for one am glad they did so.

zviratko, 2025-01-25:
> By right I meant "actually secure" rather than any moral judgement on what they're doing.

That's exactly what I meant - secure.

>In any case, Orca has (sanely IMO) just rejected Bambu's proposed change as-is, telling Bambu exactly how much their ideas area appreciated by the community. I for one am glad they did so.

I thought about it some more and they should just release something like Octoprint compatible API if they want us to use Bambu Connect, they are free to add authorization to the actual printing action as they see fit but it wouldn't cause as much friction.
It's not bad because of what it does, it is bad because it's Bambu-specific and would have to be integrated and maintained in every slicer.

jwise, 2025-01-25:
Converting this to a discussion, since I don't think there's action required from X1Plus :)