X1Plus/X1Plus #426: Images broken on install guide

candre23, 2025-01-19:
Nearly all the image links are dead on the wiki install guide:

[image]

x1-Cygnus, 2025-01-19:
yep showing the same, links dont work

bani6809, 2025-01-20:
unable to repro this. i tried from different machines, different browsers, in 3 different countries (usa, germany, uk).

duracell, 2025-01-20:
I had the same issue, after disable my ad-/js-blocker the images worked.

x1-Cygnus, 2025-01-20:
Ok cool will try
Thanks

*LIFE: A space time collision.*

aminorjourney, 2025-01-22:
Was able to download and root X1C yesterday and install X1Plus yesterday evening. :) I have a pi-hole on my home network, so it might be your specific config.

x1-Cygnus, 2025-01-22:
So what software version did you root from, I am at 1.8.2, rolled back to
1.7.5 but still had no third party button. Is it possible to root from
1.8.2 without button?  My understanding is I need firmware on machine to
support booting from micro SD for it to work. Am I mistaken?

*LIFE: A space time collision.*

jwise, 2025-01-25:
Closing cannot repro.

closed 2025-01-25 by jwise

x1-Cygnus, 2025-01-25:
Ok, no problem.

x1-Cygnus, 2025-01-25:
Ok, thanks for letting me know.