X1Plus/X1Plus #429: X1Plus Wiki - Install Guide out of date

markaudacity, 2025-01-23:
The Install Guide lists Official Rootable Firmware versions in the range 01.06.06.55 to 01.06.99.99, while X1+ v2.0 release notes say it is now based on 1.08.00.00.
It also lists 01.07.04.00 as the version to downgrade to if ORF versions do not appear in Bambu Handy; the earliest downgradable version currently listed in BH is 1.07.05.00.

psiberfunk, 2025-01-23:
There are a few misunderstandings here:

1) the firmware that x1+ overlays against is NOT the same as the bootable firmware.  Presently it overlays against the 1.08 firmware IIRC

2) Presently you need to downgrade to something in the 1.07 range to download the rootable firmware and must also be enrolled in the rootable firmware program with bamboo for your printer serial number

jheyneman, 2025-01-24:
I actually downgraded too far, to the earliest available 1.07.  I then had to upgrade back to 1.07.03 in order to be able to see the rootable 1.06 firmware after opting in.

markaudacity, 2025-01-24:
So will the correct firmware be labeled as rootable firmware, or am I just looking for a specific version number? I have now gone through enough downgrade steps to see 01.06.06.58, but it's not labeled as rootable or anything.

I am enrolled in the firmware program.

psiberfunk, 2025-01-24:
I don’t think they call it rootable on the handy app , iirc

cryptk, 2025-02-01:
As the [installation instructructions](https://github.com/X1Plus/X1Plus/wiki/Installation-Guide#unlock-your-printer-and-get-your-access-codes) say: "(it will have a version number between 01.06.06.55 and 01.06.99.99)".

The current latest official rootable firmware is 01.06.06.58