X1Plus/X1Plus #432: X1Plus and the upcoming firmware, a few questions

WimVDK, 2025-01-24:
I've followed X1Plus since its first announcement on reddit, but i wasn't willing to flash it for 2 main reasons. It's stability hadn't been widely tested publicly and i was still in my first year of printer warranty. Now that its been more tested, and i'm in my 2nd year, i'm more open to the idea. Especially based upon the latest statement of SoftFever here https://github.com/SoftFever/OrcaSlicer/pull/8103#issuecomment-2612855023 i'm definitely not updating to the latest official firmware.

I know X1Plus simplified, just replaces the stock UI. What I couldn't find an immediate answer on this. is X1Plus going to add the "authentication requirement" if for example firmware 01.09.00.00 comes out that has meaningful new features and a new X1Plus version gets released? Is it affected in any way (other than that one) with the new official firmware?

I'm probably going to flash X1Plus for now anyway just to avoid getting the firmware notification for the new official one.

RobinWhitfield, 2025-01-24:
The Discord server would be a better place to discuss this (https://discord.gg/4SfCufp6SB), but the short answer is no.

X1Plus is not simply a UI overlay, although at its core it is built on top of stock firmware. New features can be cherry picked to be implemented in new versions of X1Plus so there will never be a requirement to implement any new authentication added by Bambu Lab.

RobinWhitfield, 2025-01-24:
Not an issue.

closed 2025-01-24 by RobinWhitfield

WimVDK, 2025-01-24:
> The Discord server would be a better place to discuss this (https://discord.gg/4SfCufp6SB), but the short answer is no.

Good

> X1Plus is not simply a UI overlay, 

I know, which is why i mentioned "simplified" ;)

> although at its core it is built on top of stock firmware. New features can be cherry picked to be implemented in new versions of X1Plus so there will never be a requirement to implement any new authentication added by Bambu Lab.

Well that settles it then. Off to X1Plus this weekend. Thanks!

psiberfunk, 2025-01-24:
Yea x1+ will not be adding Bambu’s new gatekeeping “feature” .