X1Plus/X1Plus #435: USB stick support to transfer Gcode files to printer on X1Plus Extension board

Durksz, 2025-01-30:
First of all thank you for hacking the Bambulab printers and making opensource firmware for it!

Problem statement:
The company I work for, insert multionational company name here, has a policy that only encripted usb sticks are allowed.
The fact we are using micro-sd cards to feed our bambulabs at the R&D department is tolerated, for now. 
And Plugging it in to the network (Wired or wireless) is a big nono (only for updates we hook it up wifi).

So having a usb input for feeding our models into the Bambu Lab would be very very usefull

Since the Expander board has 2 USB inputs, it would be ideal for the job.

I sincerely hope you can put this feature on the roadmap for implementation in the X1Plus firmware in the near future.

RobinWhitfield, 2025-01-30:
USB sticks encrypted with what exactly?

The printer would also need to be able to handle that, which could be more difficult.

bani6809, 2025-01-30:
> The company I work for, insert multionational company name here, has a policy that only encripted usb sticks are allowed.

The possibility of support depends entirely on what encryption is used. Can you be more specific about your encryption?

Durksz, 2025-01-31:
I will try to find out what encryption is used. I Will share brand and type of usb stick we received. Once i get back in office next week.

If my memory is correct it is that reading is always permitted but writing from laptop to usb / sd is not.

Durksz, 2025-02-04:
the USB stick that is provided to us is a Apricorn NX with FIPS140-2 Level 3 validated on it.
As far as i understand it is an hardware locked usb-stick (we need to type a code to unlock it).

To be honest, being able to use an unencrypted usb stick already can be usefull, since you cannot use the micro-SD card easily anymore to transfer printfiles to the printer. When the sd card has X1Plus firmware installed on it. We have 20 or more people using the same printer....

Durksz, 2025-02-05:
I'm no proficiant programmer (mechanical engineer with some limited python / arduino programming skills), is it complicated to implement a usb-reader feature?

jwise, 2025-03-21:
USB support would be nice, but I don't think it's going to make the initial Expander release.  Leaving the bug open to track the request, though.

mattcaron, 2025-09-25:
Digging a little bit, this appears to just be a UI issue.

Plugging in a USB stick (normal, FAT, not encrypted), it gets mounted properly automagically.

```
/dev/sda1                57.7G     34.1M     57.7G   0% /media/usb0
```

But, there's no way to tell the printer to print the file on it.

mattcaron, 2025-09-25:
Created https://github.com/X1Plus/X1Plus/discussions/523 for design discussion.

When I have a PR, I'll ref this ticket.

Not sure what is needed to support encrypted USB, though...

@Durksz Do the encrypted USB sticks work on a bog standard Linux machine? Any information you can provide would be helpful.

jwise, 2026-05-16:
Dang, this is super cool.  OK I just unpacked my X1 in my basement so... this might be possible.