X1Plus/X1Plus #439: Add a way to personally set the printers access code

JeremiahSecrist, 2025-02-05:
**Is your feature request related to a problem? Please describe.**
Currently recording the access code for every printer I own is quite tedious and gets in the way of connecting to them at times. 

**Describe the solution you'd like**
It would be nice if the ui would let me set my own code that is persistent. 

**Describe alternatives you've considered**
Continue recording every code and keep switching back and forth from my password manager 

**Software Information**
 - Current X1Plus Version: 2.0 (not entirely relevant) 

**Additional context**
This is more to add convenience to lan mode.

damige, 2025-03-17:
Would be great to have this feature, as orca asks me to put this password in often.

damige, 2025-03-21:
> You can set it from command line, something like `echo -n 12345678 >/path/to/some-file` – I don't remember what the file is, though, and I can't check right now. Probably needs a reboot to take effect.

Tried that (echo -n "12345678" > /config/device/access_token) and also setting user bblp with the same pw.
After reboot the password does appear on the display, but orca still says wrong password, and mqtt cannot connect either.

Did you get it to work?

arkku, 2025-03-22:
> Did you get it to work?

Yes, I did get it to work, but now that you mention it, I'm not 100% sure whether there is also some command you need to run. Might be that I tried the file first, and it didn't work, and then I found some command in whatever script that sets it normally when you press the randomize code button.