X1Plus/X1Plus #442: <your bug name here>

W1SPS, 2025-02-19:
**Describe the bug**
X1C does not recognize the "Supertack" plate from Bambu, even though Orcaslicer had a "Nightly Release" that addressed that and added the Supertack plate to the list of bed plate suggestions. I Contacted BBL and they said this was a "Bug in their firmware". Interesting because I am using X1Plus. Is it possible X1Plus has the same bug?

**To Reproduce**
Steps to reproduce the behavior:
1. Try to print anything on the Supertack plate with said plate selected in Orcaslicer OR Bambu slicer.
2. See error

**Expected behavior**
Printer starts to calibrate and get ready to print, then tries to read the code on the Supertack plate and then returns an error that it couldn't be read. I can't just tell the printer to continue because when the plate can't be read the print nearly always fails.

**Screenshots/Photos**
Nothing to see but that error described.

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section?

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 99.00.00.00
 - AP firmware version: No idea
 - AP hardware revision: No idea
 - MC firmware version: No idea
 - MC hardware revision: No idea
 - Toolhead firmware version: No idea
 - Toolhead hardware revision: No idea
 - Computer vision model: No idea

**Is the bug present on OEM firmware?** Bambu doesn't know I'm running X1Plus, so for them so tell me there is a firmware bug, I'd say yes.
Have you tested whether this bug is also present in OEM firmware? No

W1SPS, 2025-02-19:
X1Plus version 2.0 that is

pelrun, 2025-02-19:
This is not an X1Plus bug. X1Plus is just a set of patches on top of the official firmware, so any bugs in the official firmware are likely going to be present in X1Plus as well.

As for the supertack plate - the BL product page *explicitly* states that marker detection has to be turned off to use it. There's nothing X1Plus can do about the plate having the tag printed in the wrong place.

W1SPS, 2025-02-19:
Yeah, well it didn't "explicitly state" that when I bought it. Also, how is anyone suppose to know that " X1Plus is just a set of patches on top of the official firmware" when its touted as firmware. And if I had to revert my OEM firmware back to 1.06 to install X1Plus and "X1Plus is just a set of patches on top of the official firmware" then how would I get the Supertack plate to ever work correctly if I don't plan on ever going back to the OEM firmware?

jwise, 2025-03-21:
The latest BL firmware supports Supertack.  I'll try to intercept 3.0 with this but no guarantees.  Definitely want to intercept 3.0 for filament updates, though.

W1SPS, 2025-03-31:
I'm a bit confused about how to implement that. I had to roll the BL firmware back to 01.06.0x.xx or something so I could use X1Plus. How would I implement the new BL firmware that "allegedly" fixes the supertack issue and still use X1plus?

jwise, 2025-05-11:
Punting this to 3.1, 3.0 just needs to ship

kylek29, 2025-06-18:
The X1C will recognize the build plate with the right ArUco code. I think they may have changed the ArUco code post launch, so the original stickers may not be right. Even a lot of the 3rd-party plates shipped with stickers that had the wrong code.

The way their plate identifier works is by using a camera to scan that plate code which converts to an integer (they're quite literally 1, 2, 3, ...) and that checks against the plateID  that is expected. I believe an ArUco sticker that converts to 0 will work as it's a catch-all.