X1Plus/X1Plus #452: X1Plus 2.0 release installer for MacOS x64 is actually an Apple Silicon Binary and cannot be run.

Numbski, 2025-03-21:
**Describe the bug**
Exactly as stated in the title.  It says it is an x64 binary, but it is actually arm.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the binary from the releases page.
2. Mount the DMG file.
3. Observe that it cannot be run.

**Expected behavior**
Binary should be executable.

**Screenshots/Photos**
I have a screenshot, but I see no mechanism for attaching it.

**Additional Details:**
 - Printer type X1C
 - X1Plus version: 2.0
 - AP firmware version:
 - AP hardware revision:
 - MC firmware version:
 - MC hardware revision:
 - Toolhead firmware version:
 - Toolhead hardware revision:
 - Computer vision model:

Numbski, 2025-03-21:
[image]

Numbski, 2025-03-21:
I now see this is a duplicate of #327

jwise, 2025-03-21:
Yep, dup #327.  Thanks for the report.  I'll hope not to screw this up in X1Plus 3.0.

closed 2025-03-21 by jwise