X1Plus/X1Plus #462: <your bug name here>

chuckkay, 2025-03-26:
Inconsistent device information displayed on 2 different devices .

**To Reproduce**
Go to settings-> general-> device info to see things like odometer for hours used and other information 

**Expected behavior**
Same expanded details should be visible on both machines, but i only see expanded info for one machine
[image][image][image][image]

jwise, 2025-03-26:
Interesting.  I added a bug title but this definitely seems broken.

mikefromdot, 2026-11-08:
I know this is pretty necro but by chance have any of you printed ASA/ABS on the affected printers?

I think its getting tripped up on that.

kadlerjeff, 2026-11-08:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

I've used PLA, ABS, PETG, PETG-CF,ASA-AERO. Temp of hot end and bed display
correctly on the half-ish right side of the screen. Then the 3 point list
saying preheating and bed leveling gets plopped right on-top of the graph,
that's when it stops reporting and the graph goes blank.

mikefromdot, 2026-11-08:
The graph is fixed in main, I mean the device info that has no data after Lead Screws last lubricated which is what this one is about. 

I think the issue is related to the check on ASA/ABS