X1Plus/X1Plus #471: Possible to modify printer serial number?

PatrickChenHZ, 2025-04-13:
Just wondering if it is possible through any means to modify the serial number of an X1 through X1Plus. I have no malicious intent, I happen to have a P1S that had its AP and MC board blown and also have a set of X1 AP/MC board. And I did the ungodly move of converting the P1S to an “X1S”. Wondering if it’s possible through any means to restore some cloud function by writing my P1 SN or the SN from one of my other X1 that’s always in LAN mode to it.

[image]

balika011, 2025-04-14:
Yes, but we can't tell you how.
Where did you get the x1e AP from?

PatrickChenHZ, 2025-04-14:
Thanks for confirming it is possible in theory, I will dig into the repository and figure out how. Understand why you can’t disclose the exact method but if any other more direction can be given it is well appreciated.

I have a frankenstein combo in this machine right now, AC board from P1S, one of each TH board from X1E and P1S, MC board from a X1C, AP board from X1E, button board/chamber temp sensor from X1C, hotend from P1S. They are not supposed to be compatible with each other but nothing is complaining, so I am not complaining.

To answer your question, the X1E AP board is from a X1E I maintain at work. The tool head board fried itself on the X1E and took the AP board with it. When we reached out to our reseller and they reached out to Bambu they deemed the AP board unrepairable and sent a full set of AP+TH board to replaced. But after taking the AP board out I found out it was just 2 0-ohm resistor near the USB-C that fried and one other capacitor on the other end of the board, replaced these and the board came back to life fully functional. Since the X1E have a new AP board in it now and has been working fine, I just decided to do something fun with this one.

jwise, 2025-04-14:
This is quite good.  Feel free to continue discussing it, though I am going to move it to a GitHub 'discussion' rather than an 'issue' just to keep the issues tab clear, if that's OK.