X1Plus/X1Plus #483: <your suggestion here>

WimVDK, 2025-05-19:
**Is your feature request related to a problem? Please describe.**
No. Entirely new

**Describe the solution you'd like**
Implement the updated UI (looks real good) that's present on newer official firmware

**Describe alternatives you've considered**
There are no alternatives other than the current UI

**Software Information**
 - Current X1Plus Version: latest

**Additional context**
Including the snarky tips and images :)

WolfwithSword, 2025-05-19:
Currently, updating to the new UI seems like it will require completely updating the underlying firmware stack, including going through the new authentication stuff. Right now, that doesn't quite seem feasible unless we get more volunteers to help with reverse engineering the new UI itself to perhaps backport some with or without the firmware stack (it's entirely rewritten and much harder, based on discussions done in the discord server).

Right now the focus closest to this is primarily to backport AMS 2 Pro and AMS HT support as it is slightly more feasible. However, that is also significant effort and not in the immediate priorities I believe. But we'll welcome help in any form for this.

Related: https://github.com/X1Plus/X1Plus/discussions/461 + discussions in the discord.

WimVDK, 2025-05-19:
I'd love to help but i have very limited coding skills (mostly web dev related) and zero reverse engineering skills. Other than offering to install new (beta) versions, there's not much i can do.