X1Plus/X1Plus #496: AMS update fails and hangs UI

pudrik, 2025-06-20:
**Describe the bug**
Version 3 fails to update ams to version 00.00.06.44 and hangs on second retry.

**To Reproduce**
Steps to reproduce the behavior:
Go to firmware try to update AMS from 00.00.06.40 to 00.00.06.44
Go thru all the steps to perform the update (it will fail)
then try to update again. it now gets stuck at the updating dont powercycle the printer

**Expected behavior**
Update should either succeed or fail and UI should show a OK / Cancel button depending on outcome.

**Additional Details:**
 - Printer type (X1/X1C/X1E):
 - X1Plus version: 3.0
 - AP firmware version: 00.00.32.39
 - AP hardware revision: AP05
 - MC firmware version: 00.00.27.26
 - MC hardware revision: MC07
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision: TH09
 - AMS hub firmware version: 00.00.00.42
 - AMS hub hardware revision: AHB01
 - AMS firmware version 00.00.06.40
 - AMS hardware version AMS08

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
Havent tried, dont want to revert to stock.

**Custom modifications**
-

**Additional context**
-

Shabbypenguin, 2025-06-21:
Also getting this error, it updated one of my AMS units, but trying to update the other one it keep looping like it was going to but failing to.

ronbentt, 2025-06-22:
Same issue, 2x AMS only number one updated, the second one flashed red on the feed lights on the initial install. I tried disconnecting the pair and rebooting with no ams, then plug in the second by itself to try the install and it says failed still.

pudrik, 2025-06-23:
I only have 1 ams connected atm but its identifies as my 2nd ams (AMS-2) so im not sure if its a canbus address issue or something else.

ronbentt, 2025-06-23:
My AMS #2 failed, I attempted to do the install on it solo and it still identifies as #2 so I did the Reset above and it became a single unit again. Give the reset a go as it may be the id thats throwing the whole thing.

ronbentt, 2025-06-23:
I had 2x AMS and number 2 failed. Here is how I got mine to work. 

1. Open the filament setting to access the AMS reset. 

[image]

2. With both AMS connected press the reset button

[image]

3. Turn off and disconnect all AMS, turn printer on again and connect the one that wasn't updated. It should identify as a single unit now if all things are right. Go into the update AMS firmware and update.

4. As a rule I turn the printer off, then disconnect the AMS. Start the printer and connect the AMS in the sequence you want them numbered in the daisy chain.

Marckau, 2025-07-03:
@pudrik that worked for me Thank you!

PR #507 (2025-07-14): add support for upgrading a secondary AMS
Also, use new AMS FW.

Fixes #496.
commit cb9d5b19 (2025-07-14): add support for upgrading a secondary AMS (#507)

* UpgradeDialog: correctly send command to upgrade secondary AMS
* version: upgrade to AMS 00.01.06.62

Fixes #496.
closed 2025-07-14 by jwise