X1Plus/X1Plus #525: Fresh 3.1 install falsely reports wrong peripheral versions

mattcaron, 2025-09-25:
**Describe the bug**
Every time I boot my printer, it says the peripheral board versions are wrong, but, individually, they report the correct versions.

(This is not a showstopper, it's just annoying.)

**To Reproduce**
Steps to reproduce the behavior:
1. Install 3.1 on a fresh printer.
2. Boot it.
3. Update all peripheral boards.
4. Reboot it
5. See message

**Expected behavior**
It doesn't tell me to upgrade when there's nothing to upgrade.

**Screenshots/Photos**
[image]
[image]

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 3.1
 - Base firmware version: 01.08.02.00
 - AP firmware version: 00.00.32.39
 - AP hardware revision: AP05
 - MC firmware version: 00.00.27.26
 - MC hardware revision: 
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision: TH09
 - Computer vision model: I don't know how to find this.

**Is the bug present on OEM firmware?**

No. The OEM firmware doesn't have this dialog.
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

**Custom modifications**
Add details of any software or electronic modifications you've made to your printer here.

**Additional context**
Add any other context about the problem here.

MikeOxlong83, 2025-12-05:
Also getting this problem, and when I don't get the version mismatch dialog, I get the "SD card not formatted" dialog (even though the firmware is RUNNING on the supposedly unformatted SD card)

DragonMortyBalls, 2026-04-19:
I want the newer filament database, it has the newer filaments supported for the screen, I want more color options, we're more added? We need the creator to add support for this and allow it to be updated

louhy, 2026-04-19:
Oh, yeah I don't know. As far as X1 Plus updates, yea it is tough - OSS developers do all this stuff in their free time, so it'd be rude to demand updates. 😄 One of the downsides of leaving the "official" firmware world.